Makeblock-official/Makeblock-Libraries #10: why not using the arduino library and board manager for the libraries

Cinezaster, 2016-06-07:
I just received my MegaPi. And I wanted to install the libaries and board files.

Now Arduino finally has a board manager and a library manager, would it not be useful to have makeblock libraries also on this eco system?

Cinezaster, 2016-06-07:
## Boards
Boards kan be added like:
Adafruit does it like this: https://github.com/adafruit/arduino-board-index
or 
Sparkfun: https://github.com/sparkfun/arduino_boards

## Libraries
Libraries can be added by making an issue on: https://github.com/arduino/Arduino/issues check the closed issues for how to add a new library to the ide.

This will make it a beter experience for all of your users

YanMinge, 2016-06-08:
I know this solution., and our zeropi use this solution. 
https://github.com/YanMinge/ZeroPi_package

But now more users have been accustomed to using our current solution. I will open another path to publish this solutions.  It may take a little time, Thank you for your suggestion.

closed 2016-11-10 by YanMinge