Makeblock-official/Makeblock-Libraries #11: Gyro library X and Y values are between -90 and 90

EliKrumholz, 2016-08-11:
Is there a reason why gyro.getAngleX() and gyro.getAngleY() return values between -90 and 90 while gyro.getAngleZ() returns values between -180 and 180? Why not return values between -180 and 180 for all angles?

YanMinge, 2016-08-19:
thanks for your comments. I will update next version.

clplaneguy, 2016-11-02:
I have found the example programs for the gyro module and the wall avoider.  I have enhanced both for my use.  Then I combined the two use the gyro in the wall avoider.  The example programs work exactly as expected.  But on the combination program, the gyro.getAngleZ() value is never accurate.  The gyro can not keep up if other things are happining.  

And you could return all angles between 0 and 360.

/**
 *
 *
 * @file    Ivany10.ino
 * @author  MAM
 * @version V1.0.0
 * @date    11/2/2016
 * @brief   Description: This program started with the original code for the Mbot.
 *
 * Function List:
 * 1. void MeGyro::begin(void)
 * 2. void MeGyro::update(void)
 * 3. double MeGyro::angleX(void)
 * 4. double MeGyro::angleY(void)
 * 5. double MeGyro::angleZ(void)
 *
 * \par History:
 * <pre>
 * <Author>     <Time>        <Version>      <Descr>
 * Mark Yan     2015/09/09    1.0.0          rebuild the old lib "Wall Avoider.ino"
 * MAM          10/24/2016    1.0.0          Convert to MegaPi   "Wall Avoider.ino"
 * MAM          10/25/2016    1.0.1          Increase avoidance distance
 * MAM          10/26/2016    1.0.0          File name Ivany
 * MAM          10/27/2016    1.0.0          Rename Ivany2.ino  5:01PM Remane Ivany4.ino Use Boolean switch and counter.  5:33PM Works great.  Move on to version 5 with more code. Include gyro.
 * MAM          10/27/2016    1.0.0          Saved as Ivany5.ino
 * MAM          10/31/2016    1.0.0          Saved as Ivany8.ino
 * MAM          11/1/2016     1.0.0          Printed 6 point in Notepad.  Try 4 point.  Typed in hard copy corrections.  Has been tweeked.  Works great.  Left ONLY.  Saved as Ivany9.ino
 * MAM          11/2/2016     1.0.0          Add code for 'Too close too wall, back up'.  Saved as Ivany10.ino  Now does left and right.
 * </pre>
 *
 */

#include <Arduino.h>
#include <Wire.h>
#include <SoftwareSerial.h>

#include <MeMegaPi.h>

MeUltrasonicSensor ultraSensor(PORT_7); /* Ultrasonic module can ONLY be connected to port 3, 4, 6, 7, 8 of base shield. */
MeGyro gyro;
//Encoder Motor
MeEncoderOnBoard Encoder_1(SLOT1);
MeEncoderOnBoard Encoder_2(SLOT2);

float Value;
/* boolean Right;
/*
 *MeEncoderOnBoard Encoder_3(SLOT3);
 *MeEncoderOnBoard Encoder_4(SLOT4);
*/

void isr_process_encoder1(void)                        //
{ //
  if (digitalRead(Encoder_1.getPortB()) == 0)          //
  { //
    Encoder_1.pulsePosMinus();                         //
  }                                                    //
  else {                                               //
    Encoder_1.pulsePosPlus();                          //
  }                                                    //
}                                                      //

void isr_process_encoder2(void) {                      //
  if (digitalRead(Encoder_2.getPortB()) == 0)
  { //
    Encoder_2.pulsePosMinus();                         //
[... 311 lines cut ...]
  long endTime = millis() + seconds * 1000;
  while (millis() < endTime)_loop();
}

void _loop() 

{
  Encoder_1.loop();
  Encoder_2.loop();
}

float GoodDistance() 

{
  distance  = 0;
  for (x = 1; x <= 5; x++) {

    distance = distance + ultraSensor.distanceCm();
    delay(100);
    //           Serial.print(x);
  }  //   for (x = 1; x <= 5; x++){

  distance = distance / 5;
  return distance;
}  //   float GoodDistance(){

closed 2018-06-29 by MrHezhisheng