Makeblock-official/Makeblock-Libraries #15: mBot mCore card support

benoit-pereira-da-silva, 2016-08-25:
Arduino - 1.6.8
OSX - 10.11.6 (15G31)
mBot - Blue (Bluetooth Version) with a mCore card.

1. I ve copied the files from the makeBlock Library into the app content package. (note that there is a typo error in your read me /java/ not /jave/ ) 
2. I ve restarted the Arduino app
3. The samples are available but not the card.
4. I ve have imported : #include "MeMCore.h"
4. And tryed to compile some samples  (screen shot attached) 

[image]

[image]
[image]
[image]

closed 2016-08-25 by benoit-pereira-da-silva

YanMinge, 2016-08-26:
Hi, seems it is not a compile error. if you use OSX, why you have COM1? it only will found on windows.

YanMinge, 2016-08-26:
you should select the correct port and board on tools

benoit-pereira-da-silva, 2016-08-26:
Thanks @YanMinge i have been able to fix the communication issue.