Makeblock-official/Makeblock-Libraries #26: Can't compile: 'class MeEncoderNew' has no member named 'GetCurrentSpeed'

Daemon2017, 2017-02-16:
Hello! 
Library installed as in manual, but the result is 'class MeEncoderNew' has no member named 'GetCurrentSpeed'.
[image]

dobbyzzipeng, 2017-02-17:
hello,have you included the head code?like this:
[image]

Daemon2017, 2017-02-17:
Hello!

I've tried to change, but  it doesn't help:
[image]

What else could be wrong in my code?

Daemon2017, 2017-02-17:
Hello! 

Here is a problem:
In documentation on maleblock's site: float currentSpeed = motor1.GetCurrentSpeed();
Working code: float currentSpeed = motor1.getCurrentSpeed();

So, now everything is great :)

dobbyzzipeng, 2017-02-17:
hi,you are right.Open the MeEncoderNew.cpp code and make sure the method "GetCurrentSpeed" is in it.
like this:
[image]

if not,you should update your libraries.

Makeblock-official, 2017-03-06:
You should update your libraries.

closed 2017-05-20 by YanMinge