Makeblock-official/Makeblock-Libraries #27: Ble scan support for ELET114A bluetooth module

inker610566, 2017-03-24:
I want to add feature for scanning and communicating with ble(bluetooth lower energy) device to mBot firmware.
But I can only find datasheet of its processor(BCM20710) and no way to contact module vendor.

Any hint or suggestion?

Thanks.

dobbyzzipeng, 2017-03-24:
1.use our app named makeblock that supports communication between mobile phone and mBot.
2.the ble works on unvarnished transmission mode,so you only need contact with the ble and send datas.
3.you can find the protocol in mBot firmware code.

inker610566, 2017-03-24:
Hi,
I have checked mbot_factory_firmware.ino and mbot_firmware.ino under mBlock project. Neither of them seem to used bluetooth directly. class MeBluetooth under Makeblock-Libraries seems to use UART serial interface directly. So @dobbyzzipeng you mean I can scan ble device through bluetooth UART serial interface from mbot directly? Sorry for that I don't quite familiar with how to implement ble scanner(a as central device) in an embedding system.

inker610566, 2017-03-28:
I found datasheet of [ELET114A bluetooth module](http://pan.baidu.com/s/1qW9Skd2) (in chinese) and referenced the [mcore datasheet](http://learn.makeblock.com/cn/mcore/).
According to datasheet, the RX/TX pin of mcore bluetooth module are pin 0 and pin 1.
So far I have tried writing a simple arduino program for mcore to send/receive AT command from mcore, but it failed to receive any data.
Code snippet: https://gist.github.com/inker610566/ea50fe6b7d83f7bc5fdb115eda8c6a97

inker610566, 2017-04-18:
ELET114A doesn't support scanning(as BLE central) from AT command. I will continue with mBlock android app project, that seems to be a easier way to integrate with BLE feature.

closed 2017-04-18 by inker610566

Allisterlim, 2024-08-31:
Hi, were you ever able to send .ino programs to your mbot via bluetooth? I have the mbot1 and I am trying to send programs to the mbot through bluetooth.