Makeblock-official/Makeblock-Libraries #31: mCore communicate with uno via pin11,12,13 SoftwareSerial

inker610566, 2017-05-03:
I try to send mbot data to another arduino uno board via pin 11, 12.
Here's my code on mbot:
```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(11, 12); // RX, TX
void setup()  
{
 Serial.begin(4800);
 Serial.flush();
 Serial.println("OK");
 Serial.flush();
 mySerial.begin(4800);
 mySerial.flush();
}

void loop() // run over and over
{
  if (mySerial.available()) {
    Serial.write(mySerial.read());
    Serial.flush();
  }
  if (Serial.available()) {
    mySerial.write(Serial.read());
    mySerial.flush();
  }
}
```
Here's my code on uno:
```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(12, 13); // RX, TX
void setup()  
{
 Serial.begin(4800);
 Serial.flush();
 Serial.println("OK");
 Serial.flush();
 mySerial.begin(4800);
 mySerial.flush();
}

void loop() // run over and over
{
  if (mySerial.available()) {
    Serial.write(mySerial.read());
    Serial.flush();
  }
  if (Serial.available()) {
    mySerial.write(Serial.read());
    mySerial.flush();
  }
}
```
And I connect pin11, 12 of mbot to uno pin13, 12.
But no matter what input is send from either side,
the other side only show 'OK' message at setup.

dobbyzzipeng, 2017-05-24:
1.pin 13 on uno is connected to the led. so make sure the harware issue is ok.

Makeblock-official, 2017-05-26:
Have we test it？

closed 2018-06-29 by MrHezhisheng