Makeblock-official/Makeblock-Libraries #33: Incorrect Direction Functions in starter_factory_firmware

para-programmer, 2017-05-07:
Your Forward function is actually TurnRight, Backward is actually TurnLeft, TurnLeft is Forward and TurnRight is Backward.

Relabeling the functions as above makes the Starter Kit Robot operated as intended.

See attached files with comments above appropriate functions.

[fixed_starter_factory_firmware.zip](https://github.com/Makeblock-official/Makeblock-Libraries/files/981883/fixed_starter_factory_firmware.zip)

YanMinge, 2017-05-09:
Hi, para-programmer

This may be the result of hardware connection, In order to support both Bluetooth and infrared control, we adjusted the connection.

para-programmer, 2017-05-15:
YanMinge,

Maybe you are right, but my kit was built using the instructions, and I would assume that if a connection was wrong then it would just be a matter of lefts becoming rights and forward is backward. 

I checked my connections and the only way I could get it to work was modifying the functions as stated above. This may not be an issue if I'm the only one experiencing it, but just wanted to make you aware.

closed 2018-06-29 by MrHezhisheng