Makeblock-official/Makeblock-Libraries #35: problem with block [set encoder motor on board ... rotate ... degrees at speed ... rpm]

tschmit, 2017-05-24:
using: 
* MeAuriga
* mBlock 3.4.5
* windows 10 (app installed from windows store)

[image]

It seems that the parameter `degrees` is ignored. The motor starts rotating and never stops.

YanMinge, 2017-05-24:
It seems that the feedback of the motor should not be normal. You can try to read the motor speed, if the motor speed is not properly updated, the motor connection may be faulty.

And our latest version of the mblock is 3.4.8, you can look at the update record, whether there are related defects

tschmit, 2017-05-24:
I just updated to 3.4.8 and still have the problem.
Under Linux with version 4 the bloc does not exists ? Does that mean that it will be discarded ?

tschmit, 2017-05-24:
I write the following program, and the leds stays red :(

[image]

The problems seems to be the one you point... what can I do ?

tschmit, 2017-06-02:
I finally make tests:
- with another auriga, I have the same behavior: my motor never stops
- with another motor: the another motor starts then stops after completing the required degrees.

My motors are (both !!!) defective!

YanMinge, 2017-06-13:
May be just the cable is broken.

tschmit, 2017-06-13:
unfortunately not. I use my cable with the other motor. And all run well.

YanMinge, 2017-06-16:
You can go our after-sales process, and I will close this issue.

closed 2017-09-06 by YanMinge