Makeblock-official/Makeblock-Libraries #40: Adjust to work with Arduino IDE

ajxn, 2018-03-11:
It would be great if the support for the different cards and the Library would be added in a proper way to Arduino IDE, so it will be much easier to add support for Makeblock without ruin a valid Arduino IDE installation.

That is, not suggest people to download, unzip and copy things over Arduino standard Library directory. It is a fragile way of "install" support, instead of give a proper URL in  "Settings -> Additional board URLs" and search in "Library Manager" (or add one URL).

It will be a much easier way of add Makeblock CPU boards to the Arduino IDE.

per1234, 2018-03-11:
It's pretty sad when a company making and selling Arduino products has less understanding of how the Arduino IDE works than the average beginner. It's even worse when someone is kind enough to make the necessary changes for them but they won't even make the effort to click the "Merge" button: https://github.com/Makeblock-official/Makeblock-Libraries/pull/39

YanMinge, 2018-05-18:
Previously I had a product provided arduino library as this way - (give a proper URL in "Settings -> Additional board URLs" and search in "Library Manager" (or add one URL))

https://github.com/Makeblock-official/ZeroPi_package

The reason why this method is not used for this library is that we have different products used the same chip. This may lead to confusion for some beginners.

closed 2018-08-28 by MrHezhisheng