Makeblock-official/Makeblock-Libraries #41: Motorshields peep quite a lot

nielsswinkels, 2018-05-24:
```
MeEncoderNew motor1(0x09, SLOT1);
motor1.runSpeed(100,0);

```
When running a MeEncoderNew motor at a speed lower than the max, the motor control boards give off a high pitched noise. Is it possible to get rid of this?

I suspect it to be because of the period of the PWM signal. Can that be changed somehow?

nielsswinkels, 2018-05-24:
Is there any more documentation about the different functions and how to use them?

What do these do, and how does one use them?

- setPWM
- setPulse
- setRatio

MrHezhisheng, 2018-07-05:
hello ,thanks a lot for your question.
(1) About the problem of a high pitched noise, I have communicated with the project team.
(2) You can setPWM :  motor1.setPWM (pwm_value);
                   setPulse :  motor1.setPulse (pulse_value);
                   setRatio :  motor1.setRatio (ratio_value);
You can view more details in this file: MeEncoderNew.cpp
file path: \makeblock\src\MeEncoderNew.cpp.

nielsswinkels, 2018-07-05:
Thanks for a reply.

1) We have discussed the same question here on the forum: https://forum.makeblock.com/t/changing-firmware-on-me-high-power-encoder-motor-driver/11305/9
We figured out that the reason of the high pitched noise was the frequency of the period of the pwm signal. By changing the frequency to be above human hearing range, it is unhearable. We first tried to adjust the original motordriver firmware to do this, but that was too much work. Instead we started over from scratch and implemented a new firmware for the board (with a higher frequency and pid controller) which can be found here:
https://github.com/Dealerpriest/me-high-power-motorboard-firmware

2) My question was mainly about what the functions do. What does pulse and ratio mean? But seeing as we have implemented our own firmware this question is no longer relevant for us.

closed 2018-07-05 by nielsswinkels