Makeblock-official/Makeblock-Libraries #42: RGB led handling in MegaPi firmware (and Orion, Auriga, etc.)

rdhelli, 2018-09-11:
I have succesfully connected an RGB led with Megapi and a raspberry, but the python function call from the raspberry does not light up the Leds (it works fine with other modules, like servos). I have made the following comparison. I suspect the megapi firmware is at fault, or did I do something wrong?

A. From the [web editor](http://editor.makeblock.com/ide.html#) the following script works:

```
...
MeRGBLed rgbled_6(6, 4);
...
void loop(){
    rgbled_6.setColor(1,60,0,0);
    rgbled_6.show();
}
```

B. While in PythonForMegaPi example:

in `colorloop.py`:
```
	bot = MegaPi()
	bot.start('/dev/serial0')
	bot.rgbledDisplay(6,1,1,60,0,0);
  	bot.rgbledShow(6,1);
	sleep(0.02);
```
and in `Firmware_for_MegaPi.ino`:
```
MeRGBLed led;
...
    case RGBLED:
      {
        uint8_t slot = readBuffer(7);
        uint8_t idx = readBuffer(8);
        uint8_t r = readBuffer(9);
        uint8_t g = readBuffer(10);
        uint8_t b = readBuffer(11);
        if((led.getPort() != port) || led.getSlot() != slot)
        {
          led.reset(port,slot);
        }

        if(idx>0)
        {
          led.setColorAt(idx-1,r,g,b); 
        }
        else
        {
          led.setColor(r,g,b); 
        }
        led.show();
      }
```

- [ ] **issue1**:

 the number of leds is explicitly defined in the web editor to be 4, while in the firmware, it is undefined, and so a default value of 32 is used that must be some leftover code...
(by the way in mbot firmware, it is correctly defined to be 30, but everywhere else, it is not)

- [ ] **issue2**:

 in firmware the if-else based on idx seems unnecessary, a simple `led.setColor(idx,r,g,b)` could handle the idx=0 (all leds) or values 1, 2, 3 too.

MrHezhisheng, 2018-10-08:
hi rdhelli ,your operation is correct and you should check for other problems. The parameters in this python For MegaPi example are one-to-one.
eg:
	1、bot.rgbledDisplay(6,1,1,60,0,0);
        2、Firmware_for_MegaPi.ino：
  uint8_t port = readBuffer(6); **//parameter ：6**
  case RGBLED:
  {
        uint8_t slot = readBuffer(7);**//parameter ：1**
        uint8_t idx = readBuffer(8);**//parameter ：1**
        uint8_t r = readBuffer(9);**//parameter ：60**
        uint8_t g = readBuffer(10);**//parameter ：0**
        uint8_t b = readBuffer(11);**//parameter ：0**
        if((led.getPort() != port) || led.getSlot() != slot)
        {
          led.reset(port,slot);
        }

        if(idx>0)
        {
          led.setColorAt(idx-1,r,g,b); 
        }
        else
        {
          led.setColor(r,g,b); 
        }
        led.show();
   }
   break;

rdhelli, 2018-10-08:
thanks, @MrHezhisheng 
yes, those are the parameters I meant to use. And I am out of ideas of where else to look:

- tried slot 1 or 2 (2 seem to be the default, based on experimenting with Me RJ25 and custom sensors)
- tried connecting it to ports 5, 6, 7 and 8 as well
- tried hard-coding the no. of leds in the firmware to be 4, on port 6: `MeRGBLed led(6, 4);`
- tried replacing the if-else and just use `led.setColor(idx,r,g,b)`
- and the hardware piece is not faulty, since the web editor version lit up the leds

any other ideas please?

rdhelli, 2018-11-18:
I have found the cause of this problem 
in `Firmware_for_MegaPi.ino` the led ID is defined as:
`#define RGBLED 8`

while in `megapi.py` the api is sending the values 18 and 19:
```
    def rgbledDisplay(self,port,slot,index,red,green,blue):
        self.__writePackage(bytearray([0xff,0x55,0x9,0x0,0x2,18,port,slot,index,int(red),int(green),int(blue)]))

    def rgbledShow(self,port,slot):
        self.__writePackage(bytearray([0xff,0x55,0x5,0x0,0x2,19,port,slot]))
```

solution:
modify `/examples/Firmware_for_MegaPi/Firmware_for_MegaPi.ino` to:
`#define RGBLED 18`
can you please submit this change @YanMinge ? Thanks!

(only `rgbledDisplay` is needed, `rgbledShow` and `rgbledSetColor` are obsolete/ineffective commands)

MrHezhisheng, 2018-11-26:
thanks，@rdhelli.
I will verify the problem. If there is any problem, I will submit it immediately.

MrHezhisheng, 2018-12-12:
hi @rdhelli,
I have submitted it for megapi-0.1.9, you can upgrade your megapi package.

modified:
def rgbledDisplay(self,port,slot,index,red,green,blue):
  self.__writePackage(bytearray([0xff,0x55,0x9,0x0,0x2,18,port,slot,index,int(red),int(green),int(blue)]))
to:
def rgbledDisplay(self,port,slot,index,red,green,blue):
  self.__writePackage(bytearray([0xff,0x55,0x9,0x0,0x2,0x8,port,slot,index,int(red),int(green),int(blue)]))

closed 2018-12-12 by rdhelli