Makeblock-official/Makeblock-Libraries #46: Linking error hello world app for Makeblock

basprins, 2019-01-04:
Hi,

I've just bought a ranger robot, really fun thing to have! Now it's time to rewrite the software and have it do what I want. So I downloaded the library and started with a hello world like program.

Makeblock libraries are under <arduinoinstall>/libraries/makeblock/src

First test program looks as follows:

#include <MeOrion.h>
#include <MeInfraredReceiver.h>

MeInfraredReceiver receiver(PORT6);

void setup()
{
	//receiver.begin();
	Serial.begin(9600);
}

void loop()
{
	//if (receiver.available())
	//{
	//	Serial.println("hello world");
	//}
}

But the linker is not able to do its job:

ccqa0HnK.ltrans0.ltrans.o*: In function MeSerial::available()

Error linking for board Arduino/Genuino Uno
ccqa0HnK.ltrans0.o*: (.text+0xa98): undefined reference to SoftwareSerial::available()
 
Build failed for project 'Robot'
ccqa0HnK.ltrans0.ltrans.o*: In function MeSerial::read()
ccqa0HnK.ltrans0.o*: (.text+0xac8): undefined reference to SoftwareSerial::read()
 
ccqa0HnK.ltrans0.ltrans.o*: In function MeSerial::write(unsigned char)
ccqa0HnK.ltrans0.o*: (.text+0xae8): undefined reference to SoftwareSerial::write(unsigned char)
 
ccqa0HnK.ltrans0.ltrans.o*: In function global destructors keyed to 65535_1_Robot.cpp.o.5391
ccqa0HnK.ltrans0.o*: (.text.exit+0x1c): undefined reference to SoftwareSerial::~SoftwareSerial()
 
ccqa0HnK.ltrans0.ltrans.o*: In function global constructors keyed to 65535_0_Robot.cpp.o.5389
ccqa0HnK.ltrans0.o*: (.text.startup+0x134): undefined reference to SoftwareSerial::SoftwareSerial(unsigned char, unsigned char, bool)
ccqa0HnK.ltrans0.ltrans.o*: (.rodata+0x2c): undefined reference to SoftwareSerial::peek()
ccqa0HnK.ltrans0.ltrans.o*: (.rodata+0x2e): undefined reference to SoftwareSerial::flush()
ccqa0HnK.ltrans0.ltrans.o*: (.rodata+0x3c): undefined reference to SoftwareSerial::peek()
ccqa0HnK.ltrans0.ltrans.o*: (.rodata+0x3e): undefined reference to SoftwareSerial::flush()
 
collect2.exe*: error: ld returned 1 exit status

Any help?

Thx a lot in advance

basprins, 2019-01-04:
Ah, figured that when I add an extra include it does compile + link.

#include <SoftwareSerial.h>

Why do I need to include that? Why isn't that included as part of #include <MeInfraredReceiver.h>?

MrHezhisheng, 2019-01-07:
Because SoftwareSerial is a library of the arduino platform, see: http://www.arduino.cc/en/Reference/SoftwareSerial

basprins, 2019-01-07:
Thx oopsie

closed 2019-01-07 by basprins