Makeblock-official/Makeblock-Libraries #49: This version number seems to be inconsistent.

clplaneguy, 2019-01-09:
**Is this about version number 3.2.6 of 3.26**

MrHezhisheng, 2019-01-10:
v3.26

clplaneguy, 2019-01-13:
The version numbers have seemed to make a consistent numerical progression since 2015.  If there is  a sudden vast improvement would not that be to 4.0.0 or lesser so to 3.3.0!  To jump from 3.2.5 to 3.26 just asks the question if it is a typo.  And, if this were well thought out, it would be included in the update list.

Author | Date | Version | Description
-- | -- | -- | --
Mark Yan | 2015/07/24 | 3.0.0 | Rebuild the old lib.
Rafael Lee | 2015/09/02 | 3.1.0 | Added some comments and macros.
Lawrence | 2015/09/09 | 3.2.0 | Include some Arduino's official headfiles which path specified.
Mark Yan | 2015/11/02 | 3.2.1 | fix bug on MACOS.
Mark Yan | 2016/01/21 | 3.2.2 | fix some library bugs.
Mark Yan | 2016/05/17 | 3.2.3 | add support for MegaPi and Auriga Board.
Mark Yan | 2016/07/27 | 3.2.4 | fix some JIRA issue, add PID motion for Megapi/Auriga on board encoder motor.
Mark Yan | 2018/05/16 | 3.2.5 | Correct copyright information.
**Mark Yan | 2019/01/13 | 4.0.0 | Major upgrade of all libraries**

MrHezhisheng, 2019-01-14:
Thanks, you're right. The version number is 3.2.6, and v3.26 is for display only, so it doesn't make much sense.