Makeblock-official/Makeblock-Libraries #53: Possible incorrect line in Makeblock firmware for the Ultimate 2.0 MegaPi

clplaneguy, 2019-01-28:
I am struggling to figure out the firmware for the MegaPi.  There is one line in particular that is giving me trouble and I believe is incorrect.

**2374 "str = strtok_r(0, " ", &tmp);"**

I am not familiar with this code.  But, it does not seem to match the references that I can find.

MrHezhisheng, 2019-01-28:
Please specify the problems you have encountered. What are the specific problems you have encountered with megapi?

clplaneguy, 2019-01-28:
I have the Makeblock Ultimate 2.0 (MegaPi) robot kit and am trying to
understand the firmware.  As I study and learn it, I believe I have found a
back door into the firmware.  There is a commented-out block of code in the
loop that seems to be the entry point.  When I uncomment this block of
code, it attempts to parse a command typed in.  The firmware does compile
to completion.  However, the command is not interpreted correctly.  Proper
function stops inside the subroutine "parseGcode".  It seems that the
intent of the code is to adjust the PID parameters and more.  However, they
always print out as all zeros.  This code is deep into pointers, strtok_r
and atof. All of which I am not familiar with and have not been able to
figure out.  Also, the sample code I fine on the Internet does not execute
in the Arduino IDE.

The commented out entry point is . . .

> while(Serial.available() > 0)
>   {
>     char c = Serial.read();
>    Serial.write(c);
>    buf[bufindex++]=c;
>     if((c=='\n') || (c=='#'))
>     {
>       parseCmd(buf);
>       memset(buf,0,64);
>       bufindex = 0;
>       }
>   }

Possible commands

g p 20

g I 1

g d .0.2

g p 20 I 1 d 0.2

g 1

g 2

g 3

g 4

The firmware is 2972 lines long.  Here are portions that I use as notes.

*Firmware structure*

> Loop
>     parseCmd(buf);
>     void parseCmd(char * cmd)
>         parseGcode(cmd+1);
>         void parseGcode(char * cmd)
>             g_code_cmd = str[0];
>                  if((str[0]=='P') || (str[0]=='p')){
>             else if((str[0]=='I') || (str[0]=='i')){
>             else if((str[0]=='D') || (str[0]=='d')){
>             else if((str[0]=='Z') || (str[0]=='z')){
>             else if((str[0]=='M') || (str[0]=='m')){
>                  if(g_code_cmd == '1') PID angle
>             else if(g_code_cmd == '2') PID speed
>             else if(g_code_cmd == '3') relax_angle
>             else if(g_code_cmd == '4') megapi_mode
>
>
> PID_angle.P = 20;          // 20;
> PID_angle.I = 1;           // 1;
> PID_angle.D = 0.2;         // 0.2;
> PID_speed.P = 0.06;        // 0.06
> PID_speed.I = 0.005;       // 0.005
>
> void parseCmd(char * cmd)
> {
>   if((cmd[0]=='g') || (cmd[0]=='G'))
>   {
>     // gcode
>     parseGcode(cmd+1);
>   }
> }
>
>
>
> #define SET_MEGAPI_MODE      0x12
>
>
> megapi_mode: 0
>      if(megapi_mode == BLUETOOTH_MODE)
>      if(megapi_mode == AUTOMATIC_OBSTACLE_AVOIDANCE_MODE)
>      if(megapi_mode == BALANCED_MODE
> else if(megapi_mode == LINE_FOLLOW_MODE)
>
> #define BLUETOOTH_MODE                       0x00
> #define AUTOMATIC_OBSTACLE_AVOIDANCE_MODE    0x01
> #define BALANCED_MODE                        0x02
> #define IR_REMOTE_MODE                       0x03
> #define LINE_FOLLOW_MODE                     0x04
> #define MAX_MODE                             0x05

clplaneguy, 2019-01-28:
g d 0.2
g p 20 i 1 d 0.02 1 2 3 4 p 20 p 21 i 0

clplaneguy, 2019-01-28:
My project is to run the Ultimate 2.0 balancing robot without the App.  The MegaPi board will be maintaining the balance, leaving a Raspberry Pi for other tasks.  To do this I will have to modify the firmware and establish communication with the Pi.

The firmware is providing a method to run it in interactive mode where there is a CLI.  (Not a Windows 7 CLI, but a terminal CLI.)   The more I look at and study the code, the more correct it looks.  But the DEBUG_INFO from line 2389 is all zeros even if I try to change them.

```
#ifdef DEBUG_INFO
Serial.print("PID: ");
Serial.print(p_value);
Serial.print(", ");
Serial.print(i_value);
Serial.print(",  ");
Serial.println(d_value);
#endif

```

clplaneguy, 2019-01-28:
For one thing, the token is advanced at the beginning of the loop.  It needs to be advanced at the end of the loop.

```
str = strtok_r(cmd, " ", &tmp);
while(str!=NULL)
  {
  if((str[0]=='P') || (str[0]=='p')){
    p_value = atof(str+1);
    }
  else if((str[0]=='I') || (str[0]=='i')){
    i_value = atof(str+1);
    }
   else if((str[0]=='D') || (str[0]=='d')){
    d_value = atof(str+1);
    }
   else if((str[0]=='Z') || (str[0]=='z')){
    relax_angle  = atof(str+1);
    }
    else if((str[0]=='M') || (str[0]=='m')){
     megapi_mode  = atof(str+1);
     }
  **STR = STRTOK_R(0, " ", &TMP);**
  }

```
For another thing, the corrected code parses "g p 1.5" as
1 g
2 p
3 1
4 5
The documentation I can find, say that the second parameter of the subroutine call should be a pointer to the delimiter, not "" "".(NOT THE DELIMITER ITSELF)

clplaneguy, 2019-01-29:
atof is not working as expected.

clplaneguy, 2019-01-29:
The atof command was not working. I had to change the indexing and now it works fine.  As a novice, I find it difficult to believe that I am improving Makeblock's firmware.
```
/**
 * \par Function
 *    parseGcode
 * \par Description
 *    The function used to configure parameters for balance car.
 * \param[in]
 *    cmd - Gcode command
 * \par Output
 *    None
 * \return
 *    None
 * \par Others
 *    None
 */
void parseGcode(char * cmd)
{
  char * tmp;
  char * str;
  char g_code_cmd;
  str = strtok_r(cmd, " ", &tmp);
  g_code_cmd = str[0];
  Serial.println("----------------------------------------------------------");
  Serial.println("Inside parseGcode()");
  while(str!=NULL)
  {
    Serial.println("Inside 'while(str!=NULL'");
    Serial.print  ("str[0]");
    Serial.println(str[0]);
    Serial.print  ("g_code_cmd is '");
    Serial.print  (g_code_cmd);
    Serial.println("'");
    if((str[0]=='P') || (str[0]=='p')){
      Serial.print  ("1) p ");
      str = strtok_r(0, " ", &tmp);
      p_value = atof(str);
      Serial.println(p_value);
    }else if((str[0]=='I') || (str[0]=='i')){
      Serial.print  ("2) I ");
      str = strtok_r(0, " ", &tmp);
      i_value = atof(str);
      Serial.println(i_value);
    }else if((str[0]=='D') || (str[0]=='d')){
      Serial.print  ("3) d ");
      str = strtok_r(0, " ", &tmp);
      d_value = atof(str);
      Serial.println(d_value);
    }
    else if((str[0]=='Z') || (str[0]=='z')){
      str = strtok_r(0, " ", &tmp);
      relax_angle  = atof(str);
    }
    else if((str[0]=='M') || (str[0]=='m')){
      str = strtok_r(0, " ", &tmp);
      megapi_mode  = atof(str);
    }
  str = strtok_r(0, " ", &tmp);
  }
  Serial.print  ("PID: ");
  Serial.println(p_value);
//#ifdef DEBUG_INFO
  Serial.print  ("temp P:");
  Serial.print  (p_value);
  Serial.print  (", temp I:");
  Serial.print  (i_value);
  Serial.print  (", temp D:");
  Serial.print  (d_value);
  Serial.print  (", relax angle:");
  Serial.print  (relax_angle);
  Serial.print  (", megapi mode:");
  Serial.println(megapi_mode);
  Serial.print  ("g_code_cmd is: ");
  Serial.println(g_code_cmd);
//#endif
  if(g_code_cmd == '1')
  {
    Serial.println("Adjust angle PID");
    PID_angle.P = p_value;
    PID_angle.I = i_value;
    PID_angle.D = d_value;
    EEPROM.put(BALANCED_CAR_ANGLE_PID_ADDR, PID_angle.P);
    EEPROM.put(BALANCED_CAR_ANGLE_PID_ADDR+4, PID_angle.I);
    EEPROM.put(BALANCED_CAR_ANGLE_PID_ADDR+8, PID_angle.D);
  }
  else if(g_code_cmd == '2')
  {
    Serial.println("Adjust speed PID");
    PID_speed.P = p_value;
    PID_speed.I = i_value;
    PID_speed.D = d_value;
    EEPROM.put(BALANCED_CAR_SPEED_PID_ADDR, PID_speed.P);
    EEPROM.put(BALANCED_CAR_SPEED_PID_ADDR+4, PID_speed.I);
    EEPROM.put(BALANCED_CAR_SPEED_PID_ADDR+8, PID_speed.D);
  }
  else if(g_code_cmd == '3')
  {
    Serial.println("Set to relax_angle");
    RELAX_ANGLE = relax_angle;
    EEPROM.put(BALANCED_CAR_NATURAL_BALANCE, relax_angle);
  }
  else if(g_code_cmd == '4')
  {
    if(EEPROM.read(MEGAPI_MODE_CONFIGURE) != megapi_mode)
    {
      EEPROM.write(MEGAPI_MODE_CONFIGURE, megapi_mode);
    }
    Serial.print("megapi_mode: ");
    Serial.println(megapi_mode);
  }
}

```

clplaneguy, 2019-01-29:
The MegaPi firmware has a manual Terminal CLI interface that is completely undocumented.  In fact, it seems to be most of the code.  Or at least has access to most of the code.  There is an app for the MegaPi that I do not have access to and have never used.  It could be that this code is for the app and the CLI gives terminal access to the routines. Although it seems to be a draft or has not keep up with changes in the strtok_r and atof commands.  I have figured out enough to use the firmware code to balance my robot.  Makeblock's firmware balance code is more complicated than my balance code.  But it may work better.  Now I will see if I can get a Bluetooth or 2.4G serial connection.

closed 2019-02-01 by MrHezhisheng