Makeblock-official/Makeblock-Libraries #54: Add method for bluetooth module (not MeBluetooth) for MegaPi.

clplaneguy, 2019-01-30:
I have the Ultimate 2.0 and would like to use the included bluetooth module to connect to my PC.  I am not finding an example or even the code to do so.  Is there a plan to improve and document this.

MrHezhisheng, 2019-02-01:
1、We have a bluetooth adapter to connect the Megapi and PC, and no related libraries have been developed yet.  About  bluetooth adapter ,you can see: [https://item.jd.com/25542527466.html?jd_pop=ecad6ed0-2d06-4300-9687-94d526fd8386](url).

2、Thank you for asking so many questions, but there are too many for us to answer one by one.  It's going to be difficult. Can you discuss with me on this issue in the near future?Don't open too many issues.

MrHezhisheng, 2019-02-02:
[https://www.makeblock.com/cn/mbot/210006.html](url)