Makeblock-official/Makeblock-Libraries #55: Utrasonic

quimif, 2019-01-31:
The program of Utrasonic isnt working.Can someone please send a good program? Im still a student.

MrHezhisheng, 2019-02-01:
You can test this code in Orion.See: [https://github.com/Makeblock-official/Makeblock-Libraries/tree/master/examples/Me_UltrasonicSensor/UltrasonicSensorTest](url)

MrHezhisheng, 2019-02-01:
If not, please change your module.

quimif, 2019-02-02:
Im using that  https://github.com/Makeblock-official/Makeblock-Libraries/tree/master/examples/Me_UltrasonicSensor/UltrasonicSensorTest but still nothing

MrHezhisheng, 2019-02-11:
I can test, you find the problem yourself.

vankena, 2019-03-20:
I meet the same problem on Ultimate 2.0. Arduino IDE shows the error information:
`D:\Arduino\libraries\makeblock\src\MeSuperVariable.cpp: In member function 'void* MeSuperVariable::set_variable(char*, uint8_t, void*, uint8_t)':

D:\Arduino\libraries\makeblock\src\MeSuperVariable.cpp:241:14: warning: converting 'false' to pointer type 'void*' [-Wconversion-null]`
@MrHezhisheng Do you mind giving me some sort of a clue ?

MrHezhisheng, 2019-03-21:
Which program are you running?

MrHezhisheng, 2019-03-21:
@vankena  Can you send me your program?

vankena, 2019-03-22:
> @vankena Can you send me your program?

 https://github.com/Makeblock-official/Makeblock-Libraries/tree/master/examples/Me_UltrasonicSensor/UltrasonicSensorTest

MrHezhisheng, 2019-03-25:
@vankena  I suggest you reinstall your Arduino IDE.

vankena, 2019-03-25:
> @vankena I suggest you reinstall your Arduino IDE.

I find the problem that I just forgot to replace 'MeOrion.h' with 'MeMegaPi.h'. Sorry to bother you.