Makeblock-official/Makeblock-Libraries #70: M1, M2 motors on Auriga

king-jann, 2021-01-23:
Changing Orion to Auriga on DC motor driving examples does nothing.
What changes need to be made to send signal to M1 / M2

jmchiappa, 2021-01-25:
@king-jann,

you can drive directly output pins by using this sketch. For more information, please refer to the TB6612 datasheet.
Do not forget to select the right target ATMega2560 before compiling.

#include "Arduino.h"

#define INA1  48
#define INA2  49
#define PWMA  11

#define INB1  47
#define INB2  46
#define PWMB  10

void Forward_M1(int speed) {
  digitalWrite(INA1,HIGH);
  digitalWrite(INA2,LOW);
  analogWrite(PWMA,constrain(speed,0,255));
}

void Forward_M2(int speed) {
  digitalWrite(INB1,HIGH);
  digitalWrite(INB2,LOW);
  analogWrite(PWMB,constrain(speed,0,255));
}

void Backward_M1(int speed) {
  digitalWrite(INA1,LOW);
  digitalWrite(INA2,HIGH);
  analogWrite(PWMA,constrain(speed,0,255));
}

void Backward_M2(int speed) {
  digitalWrite(INB1,LOW);
  digitalWrite(INB2,HIGH);
  analogWrite(PWMB,constrain(speed,0,255));
}

void Stop_M1(void) {
  digitalWrite(INA1,LOW);
  digitalWrite(INA2,LOW);  
}

void Stop_M2(void) {
  digitalWrite(INB1,LOW);
  digitalWrite(INB2,LOW);  
}

void setup() {
  // io init
  pinMode(INA1,OUTPUT);
  pinMode(INA2,OUTPUT);
  pinMode(INB1,OUTPUT);
  pinMode(INB2,OUTPUT);
}

void loop() {
  Forward_M1(100);
  Forward_M2(100);
  delay(3000);
  Backward_M1(100);
  Backward_M2(100);
  delay(3000);
  Stop_M1();
  Stop_M2();
  delay(1000);
}

Hope it helps.

nbourre, 2022-10-03:
I also have this problem with the Me_EncoderDriver, but using the examples under Me_OnBoard_Encoder works.

E.g. :
```cpp
#include <MeAuriga.h>

MeEncoderOnBoard Encoder_1(SLOT1);
MeEncoderOnBoard Encoder_2(SLOT2);

void isr_process_encoder1(void)
{
  if(digitalRead(Encoder_1.getPortB()) == 0)
  {
    Encoder_1.pulsePosMinus();
  }
  else
  {
    Encoder_1.pulsePosPlus();;
  }
}

void isr_process_encoder2(void)
{
  if(digitalRead(Encoder_2.getPortB()) == 0)
  {
    Encoder_2.pulsePosMinus();
  }
  else
  {
    Encoder_2.pulsePosPlus();
  }
}

void setup()
{
  attachInterrupt(Encoder_1.getIntNum(), isr_process_encoder1, RISING);
  attachInterrupt(Encoder_2.getIntNum(), isr_process_encoder2, RISING);
  Serial.begin(115200);
  
  //Set PWM 8KHz
  TCCR1A = _BV(WGM10);
  TCCR1B = _BV(CS11) | _BV(WGM12);

  TCCR2A = _BV(WGM21) | _BV(WGM20);
  TCCR2B = _BV(CS21);
}

void loop()
{
  if(Serial.available())
  {
    char a = Serial.read();
    switch(a)
    {
      case '0':
      Encoder_1.setTarPWM(0);
      Encoder_2.setTarPWM(0);
      break;
      case '1':
      Encoder_1.setTarPWM(100);
      Encoder_2.setTarPWM(-100);
      break;
      case '2':
      Encoder_1.setTarPWM(200);
      Encoder_2.setTarPWM(-200);
      break;
      case '3':
      Encoder_1.setTarPWM(255);
      Encoder_2.setTarPWM(-255);
      break;
      case '4':
      Encoder_1.setTarPWM(-100);
      Encoder_2.setTarPWM(100);
      break;
      case '5':
      Encoder_1.setTarPWM(-200);
      Encoder_2.setTarPWM(200);
      break;
      case '6':
      Encoder_1.setTarPWM(-255);
      Encoder_2.setTarPWM(255);
      break;
      default:
      break;
    }
  }
  Encoder_1.loop();
  Encoder_2.loop();
  Serial.print("Spped 1:");
  Serial.print(Encoder_1.getCurrentSpeed());
  Serial.print(" ,Spped 2:");
  Serial.println(Encoder_2.getCurrentSpeed());
}
```