Makeblock-official/Makeblock-Libraries #73: mBot2 Error - AttributeError: '__CyberPi' object has no attribute '_protocol'. Did you mean: 'protocol'?

jiapei100, 2022-09-24:
```console
➜  ~ python
Python 3.10.6 (main, Aug 10 2022, 11:40:04) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import makeblock
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "~/.local/lib/python3.10/site-packages/makeblock/__init__.py", line 35, in <module>
    from .boards import *
  File "~/.local/lib/python3.10/site-packages/makeblock/boards/__init__.py", line 21, in <module>
    cyberpi = CyberPi.__CyberPi().api
  File "~/.local/lib/python3.10/site-packages/makeblock/boards/cyberpi/__init__.py", line 35, in __init__
    api_cyberpi_api.autoconnect()
  File "~/.local/lib/python3.10/site-packages/makeblock/modules/cyberpi/api_cyberpi_api.py", line 16, in autoconnect
    module_auto = BaseModuleAuto(board)
  File "~/.local/lib/python3.10/site-packages/makeblock/modules/cyberpi/modules.py", line 10, in __init__
    self.setup(board,index)
  File "~/.local/lib/python3.10/site-packages/makeblock/modules/cyberpi/modules.py", line 16, in setup
    board._autoconnect()
  File "~/.local/lib/python3.10/site-packages/makeblock/boards/base/__init__.py", line 44, in _autoconnect
    self.broadcast()
  File "~/.local/lib/python3.10/site-packages/makeblock/boards/cyberpi/__init__.py", line 55, in broadcast
    while not self.protocol.ready:
  File "~/.local/lib/python3.10/site-packages/makeblock/boards/base/__init__.py", line 137, in protocol
    return self._protocol
AttributeError: '__CyberPi' object has no attribute '_protocol'. Did you mean: 'protocol'?
```

joffyy, 2023-02-06:
> Also encountering this error.

Have any of y'all resolved this issue? I'm currently facing this issue my self.

solan30003, 2023-05-05:
I also faced this problem(python:3.9.16).Hope MakeBlock company technical personnel to respond.

b-paquette, 2024-12-17:
This happens if you don't have anything connected to the raspberry pi