Makeblock-official/Makeblock-Libraries #74: Unable to compile with "MeAuriga.h"

nbourre, 2022-09-26:
Using Arduino IDE 2.0, I'm trying to compile code using "MeAuriga.h" library, but I always get the error :

```
my_user_path\Documents\Arduino\libraries\Makeblock-Libraries-3.26\src\MeSuperVariable.cpp: In member function 'void* MeSuperVariable::set_variable(char*, uint8_t, void*, uint8_t)':
my_user_path\Documents\Arduino\libraries\Makeblock-Libraries-3.26\src\MeSuperVariable.cpp:241:14: error: cannot convert 'bool' to 'void*' in return
```

Here's a basic example which does not compile.

```cpp
#include <MeAuriga.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("test");
  delay(1000);
}
```

nbourre, 2022-09-26:
Please check this PR : https://github.com/Makeblock-official/Makeblock-Libraries/pull/60

DYy2021, 2022-09-27:
Your problem is caused by the return value of the 'set_variable' function in the MeSuperVariable.cpp file in the Makeblock-Libraries-3.26 library, there are now two solutions, 1. Libraries latest library is no MeSuperVariable.cpp file, so this problem is not a problem in the latest library, the new library address is https://github.com/Makeblock-official/Makeblock-Libraries.
2. You need to change the 'set_variable' function in the MeSuperVariable.cpp file from 'return false;' to 'return;' in line 241 to save it without error. See the modified look in the image below.
[image]

nbourre, 2022-09-27:
I have already fixed the problem by myself, but the latest release contains the bug.

Also, there are more than a return with a Boolean in the function.